kiibohd/configurator #210: AppImage Install Crashing due to compositor

Nalum, 2022-04-27:
I've installed the latest release and am having the following error message when trying to run:

```
[2290271:0427/180629.057628:FATAL:gpu_data_manager_impl_private.cc(1034)] The display compositor is frequently crashing. Goodbye.
```

chris-aeviator, 2022-06-17:
I have the same problem with a fresh arch installation. Why no one cares for the OP since april? Isnt this a commercial company behind this repo?

chris-aeviator, 2022-06-17:
Running into the same issue when using the 'other linux" build.

```
(electron) The default value of app.allowRendererProcessReuse is deprecated, it is currently "false".  It will change to be "true" in Electron 9.  For more information please check https://github.com/electron/electron/issues/18397
(node:72799) DeprecationWarning: Passing functions, DOM objects and other non-cloneable JavaScript objects to IPC methods is deprecated and will throw an exception beginning with Electron 9.
(node:72799) DeprecationWarning: Passing functions, DOM objects and other non-cloneable JavaScript objects to IPC methods is deprecated and will throw an exception beginning with Electron 9.
[72799:0617/073254.011289:FATAL:gpu_data_manager_impl_private.cc(1034)] The display compositor is frequently crashing. Goodbye.
[1]    72799 trace trap (core dumped)  ./kiibohd-configurator
```

superboot, 2023-06-18:
I have the same issue on Fedora 37.
```
(electron) The default value of app.allowRendererProcessReuse is deprecated, it is currently "false".  It will change to be "true" in Electron 9.  For more information please check https://github.com/electron/electron/issues/18397
[1136572:0618/104908.318634:FATAL:gpu_data_manager_impl_private.cc(1034)] The display compositor is frequently crashing. Goodbye.
```
What can be done?

superboot, 2023-06-19:
I found a work around mentioned on the LosslessCut github project that had the same error.

Add the ```--no-sandbox``` option to the app-image command, and it works.

[Reference](https://github.com/mifi/lossless-cut/issues/1025)

haata, 2023-06-19:
Thanks for the solution!
I haven't had the same issues on my Arch Linux install.

closed 2023-07-18 by Nalum