kiibohd/configurator #222: Flashing Infinity Ergodox with Linux App-Image Configurator results in white screen

superboot, 2023-06-20:
After clicking the "flash" button, and the software prompting me to press the button on the bottom of the ergodox, the LCD on the ergodox turns red (as it is supposed to in flashing mode), but the configurator's interface screen goes all white, but for the "file menu" at the top. 

What can I try, in order to fix this, and flash my keyboard?

chaimleib, 2023-06-26:
I was having trouble flashing my ergodox infinity for the same reason. On arch linux, my workaround was:

1.
```
sudo pacman -S --needed dfu-util
```

2. In the configurator, click "Flash", but do not put the keyboard in flash mode yet. There should be a red message that dfu-util could not find any dfu devices. But this step compiles the firmware.
3.  Note the location of the .bin firmware file(s).
4. To prevent white screen when we enter flash mode, go back.
5. Put the keyboard in flash mode.
6.  The leading comma in the arg to -d is specified in the manpages, don't leave it out. On a split keyboard, if the right side screen turns red, flash the right side firmware; if the left turns red, flash the left side firmware. After flashing the screen should return to normal color.
```
sudo dfu-util -d ,$(lsusb | awk '/Ergo/ { print $6 }') -D path/to/firmware.bin
```
7. If you have a split keyboard and want to flash the other side, swap the USB cables and repeat from step 5.
8. If after flashing your keyboard does not respond, try running `lsusb` to force re-enumeration of the USB devices.

bendangelo, 2024-05-28:
I have the same issue with the Whitefox keyboard. I'm on Popos.